Chill-Division/M5Stack-ESPHome #1: Not working

Cres09, 2023-04-25:
Hellow,

First of all sorry for my bad english.

I'm trying to make work de Thermal Camera Unit (MLX90640). when I hit compile in esphome on home assistant the log says the following:

> INFO Reading configuration /config/esphome/thermal-camera.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing espatom-thermalcamera (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- Wire @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/camera_mlx90640/camera_mlx90640.cpp.o
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/camera_mlx90640/mlx90640_image.cpp.o
src/esphome/components/camera_mlx90640/mlx90640_image.cpp:1:10: fatal error: FS/SPIFFS.h: No such file or directory
 #include "FS/SPIFFS.h"
          ^~~~~~~~~~~~~
compilation terminated.
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/json/json_util.cpp.o
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/logger/logger.cpp.o
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/md5/md5.cpp.o
*** [/data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/camera_mlx90640/mlx90640_image.cpp.o] Error 1
Compiling /data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/mdns/mdns_component.cpp.o
In file included from src/esphome/components/camera_mlx90640/camera_mlx90640.cpp:1:
src/esphome/components/camera_mlx90640/camera_mlx90640.h:7:10: fatal error: esphome/components/text_sensor/text_sensor.h: No such file or directory
 #include "esphome/components/text_sensor/text_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/camera_mlx90640/camera_mlx90640.cpp.o] Error 1
========================== [FAILED] Took 3.13 seconds ==========================

My configuration:

> `esphome:
  name: "espatom-thermalcamera"
  libraries:
  - "Wire"
  - "FS"
  - "SPIFFS"

esp32:
  board: esp32dev
  framework:
    type: arduino

Enable logging
logger:

Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

   Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermal-Camera Fallback Hotspot"
    password: 

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/chill-Division/M5Stack-ESPHome/
      ref: main
    components: camera_mlx90640

sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 26  # I2C SDA PIN
  scl: 32  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 15 # Minimal temperature for color mapping
  maxtemp: 40 # Maximal temperature for color mapping
  refresh_rate: 0x04 # 0x05 For 16Hz or 0x04 for 8Hz
  min_temperature:
      name: "MLX90640 Min temp"
  max_temperature:
      name: "MLX90640 Max temp"
  mean_temperature:
      name: "MLX90640 Mean temp"
  median_temperature:
      name: "MLX90640 Median temperature"

web_server:
  port: 80`

I suppose i'm making some kind of mistake but i have been not abble to debug it, i'll apreciate some help.
[logs_thermal-camera_compile (1).txt](https://github.com/Chill-Division/M5Stack-ESPHome/files/11325747/logs_thermal-camera_compile.1.txt)

ChillingSilence, 2023-04-25:
Make sure you have the included libraries that the top of the MLX90640 page shows (Naturally you can change your esphome device name though):
```
esphome:
  name: "espatom-ir2"
  libraries:
    - SPI
    - Wire
    - adafruit/Adafruit BusIO @ 1.9.6
```

You'll also want to go to the main ESPHome page, and clean the build files for your device if you've been playing around with other things previously, by tapping on the 3-dot menu, and choosing "Clean build files":
[image]

Let us know how you go :)

Cres09, 2023-04-26:
Thank you very much for your response. It is not working, now it says the following:

src/esphome/components/camera_mlx90640/camera_mlx90640.h:7:10: fatal error: esphome/components/text_sensor/text_sensor.h: No such file or directory
 #include "esphome/components/text_sensor/text_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/espatom-thermalcamera/.pioenvs/espatom-thermalcamera/src/esphome/components/camera_mlx90640/camera_mlx90640.cpp.o] Error 1
========================== [FAILED] Took 7.68 seconds ==========================`
[logs_thermal-camera_run.txt](https://github.com/Chill-Division/M5Stack-ESPHome/files/11335944/logs_thermal-camera_run.txt)

ChillingSilence, 2023-04-26:
What was it you've changed? Did you include those libraries, or clean your build files?

Cres09, 2023-04-26:
sorry. Yes I clean de build files and included de libraries. Here are the code:
"esphome:
  name: "espatom-thermalcamera"
  libraries:
    - SPI
    - Wire
    - adafruit/Adafruit BusIO @ 1.9.6
    - SPIFFS
    - FS

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermal-Camera Fallback Hotspot"
    password: 

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/chill-Division/M5Stack-ESPHome/
      ref: main
    components: camera_mlx90640

sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 26  # I2C SDA PIN
  scl: 32  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 15 # Minimal temperature for color mapping
  maxtemp: 40 # Maximal temperature for color mapping
  refresh_rate: 0x04 # 0x05 For 16Hz or 0x04 for 8Hz
  min_temperature:
      name: "MLX90640 Min temp"
  max_temperature:
      name: "MLX90640 Max temp"
  mean_temperature:
      name: "MLX90640 Mean temp"
  median_temperature:
      name: "MLX90640 Median temperature"

web_server:
  port: 80"

ChillingSilence, 2023-04-26:
Try 3x backticks ( ` ) for "code".

My apologies, I had misunderstood and thought you were using the MLX90614. You were right with only requiring:
```
esphome:
  name: "espatom-thermalcamera"
  libraries:
    - FS
    - Wire
    - SPIFFS
```

Looks like there may be an issue with inclusion of text_sensor.h, I'll get onto that now.

JakeTheRabbit, 2023-04-26:
This works for me: 

`esphome:
  name: espatom-mlx90640
  friendly_name: espatom-mlx90640
  includes:
  libraries:
    - Wire
    - FS
    - SPIFFS
  
external_components:
  - source: 
      type: local
      path: components
    components: [camera_mlx90640]

esp32:
  board: m5stack-core-esp32
  framework:
    type: arduino
logger:
api:
  encryption:
    key: 
ota:
  password: 
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: Espatom-Mlx90640
    password: Tz7rmsB41aMv
captive_portal:
text_sensor: 
sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 26  # I2C SDA PIN
  scl: 32  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 24 # Minimal temperature for color mapping
  maxtemp: 35 # Maximal temperature for color mapping
  refresh_rate: 0x05 # For 16Hz or 0x04 for 8Hz
  filter_level: 10.0 # Level in degree of the filter. If there a difference above the level, the pixel is approximated by the average of the next two pixels
  min_temperature:
      name: "MLX90640 Min temp"
  max_temperature:
      name: "MLX90640 Max temp"
  mean_temperature:
      name: "MLX90640 Mean temp"
  median_temperature:
      name: "MLX90640 Median temperature"

    #sensors:
web_server:
  port: 80`

Using m5stack atom lite and m5stack mlx90640 version 1. It won't work for version 2.

ChillingSilence, 2023-04-26:
Could you try that @Cres09 ? Just grabbing the components/camera_mlx90640 folder and putting it into your ~/config/esphome/components directory, and then instead of using the online external components try:
```
external_components:
  - source: 
      type: local
      path: components
    components: [camera_mlx90640]
```

JakeTheRabbit, 2023-04-29:
I have just done a test I compiled a new ESP Atom device and it worked and compiled. I have attached the yaml I used and the components I put in the  \config\esphome\components folder. 

My method is to create a device in esphome but not connect to the physical esp32 just create the file so it is named nicely. Then I edit the yaml in that file with the below config.  Then I install using manual install and download the file instead of trying to flash. Then go to https://web.esphome.io/ and do a manual install uploading the file there. I find this way it doesn't fuck out. 

```
esphome:
  name: espatom-mlx90640-2
  friendly_name: espatom-mlx90640-2
  includes:
  libraries:
    - Wire
    - FS
    - SPIFFS

external_components:
  - source: 
      type: local
      path: components
    components: [camera_mlx90640]

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:

    key: "your key"

ota:
  password: "your pw"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espatom-Mlx90640-2"
    password: "uYOjolJyOZSj"

captive_portal:
    
text_sensor: 
sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 2  # I2C SDA PIN
  scl: 1  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 24 # Minimal temperature for color mapping
  maxtemp: 35 # Maximal temperature for color mapping
  refresh_rate: 0x05 # For 16Hz or 0x04 for 8Hz
  filter_level: 10.0 # Level in degree of the filter. If there a difference above the level, the pixel is approximated by the average of the next two pixels
  min_temperature:
      name: "MLX90640 Min temp 2"
  max_temperature:
      name: "MLX90640 Max temp 2"
  mean_temperature:
      name: "MLX90640 Mean temp 2"
  median_temperature:
      name: "MLX90640 Median temperature 2"

    #sensors:
web_server:
  port: 80
```

[components.zip](https://github.com/Chill-Division/M5Stack-ESPHome/files/11359052/components.zip)

JakeTheRabbit, 2023-04-29:
I have just done a test I compiled a new ESP Atom device and it worked and compiled. I have attached the yaml I used and the components I put in the  \config\esphome\components folder. 

My method is to create a device in esphome but don't actually add one so you get the nice name. Then edit the device with the yaml and install and dont actually install to the device just download manually. Then go to https://web.esphome.io/ and do a manual install uploading the file there. I find this way it doesn't fuck out. 

```
esphome:
  name: espatom-mlx90640-2
  friendly_name: espatom-mlx90640-2
  includes:
  libraries:
    - Wire
    - FS
    - SPIFFS

external_components:
  - source: 
      type: local
      path: components
    components: [camera_mlx90640]

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:

    key: "your key"

ota:
  password: "your pw"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espatom-Mlx90640-2"
    password: "uYOjolJyOZSj"

captive_portal:
    
text_sensor: 
sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 2  # I2C SDA PIN
  scl: 1  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 24 # Minimal temperature for color mapping
  maxtemp: 35 # Maximal temperature for color mapping
  refresh_rate: 0x05 # For 16Hz or 0x04 for 8Hz
  filter_level: 10.0 # Level in degree of the filter. If there a difference above the level, the pixel is approximated by the average of the next two pixels
  min_temperature:
      name: "MLX90640 Min temp 2"
  max_temperature:
      name: "MLX90640 Max temp 2"
  mean_temperature:
      name: "MLX90640 Mean temp 2"
  median_temperature:
      name: "MLX90640 Median temperature 2"

    #sensors:
web_server:
  port: 80
```

[components.zip](https://github.com/Chill-Division/M5Stack-ESPHome/files/11359052/components.zip)

Obviously change the pins and the board type. I find if I have the wrong board type it won't work sometimes.

Cres09, 2023-04-30:
I finally was able to compile and upload the sketch to my board. What I was missing was to add "text_sensor:" when added it, it worked properly.

Now I'm facing an other problem. At first starts giving parameters but with an error saying "[E][MLX90640:126]: Parameter extraction failed", then stops working and say "[E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1"

The code:
```esphome:
  name: "espatom-thermalcamera"
  includes:
  libraries:
    - Wire
    - FS
    - SPIFFS

external_components:
  - source: 
      type: local
      path: components
    components: [camera_mlx90640]

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Thermal-Camera Fallback Hotspot"
    password: 

captive_portal:

text_sensor: 
sensor:
camera_mlx90640:
  id: thermal_cam
  update_interval: 5s
  sda: 21  # I2C SDA PIN
  scl: 22  # I2C SCL PIN
  frequency: 400000  # I2C Clock Frequency
  address: 0x33 # MLX90640 Address
  mintemp: 15 # Minimal temperature for color mapping
  maxtemp: 40 # Maximal temperature for color mapping
  refresh_rate: 0x04 # 0x05 For 16Hz or 0x04 for 8Hz
  filter_level: 10.0
  min_temperature:
      name: "MLX90640 Min temp"
  max_temperature:
      name: "MLX90640 Max temp"
  mean_temperature:
      name: "MLX90640 Mean temp"
  median_temperature:
      name: "MLX90640 Median temperature"

web_server:
  port: 8080

Logs:
[D][sensor:110]: 'MLX90640 Min temp': Sending state 40.00000 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Max temp': Sending state 171.64307 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Mean temp': Sending state 155.59184 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Median temperature': Sending state 154.83989 °C with 2 decimals of accuracy
[ 10791][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 10792][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 10795][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 10803][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[I][MLX90640:274]: Min temperature : 23.71 C 
[I][MLX90640:275]: Max temperature : 30.91 C 
[I][MLX90640:276]: Mean temperature : 28.77 C 
[I][MLX90640:277]: Median temperature : 29.31 C 
[D][sensor:110]: 'MLX90640 Min temp': Sending state 23.71209 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Max temp': Sending state 30.91162 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Mean temp': Sending state 28.76557 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Median temperature': Sending state 29.30715 °C with 2 decimals of accuracy
[E][MLX90640:271]: MLX READING VALUE ERRORS
[I][MLX90640:274]: Min temperature : 23.92 C 
[I][MLX90640:275]: Max temperature : 39.09 C 
[I][MLX90640:276]: Mean temperature : nan C 
[I][MLX90640:277]: Median temperature : 29.27 C 
[D][sensor:110]: 'MLX90640 Min temp': Sending state 23.92449 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Max temp': Sending state 39.08709 °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Mean temp': Sending state nan °C with 2 decimals of accuracy
[D][sensor:110]: 'MLX90640 Median temperature': Sending state 29.26779 °C with 2 decimals of accuracy
[ 25766][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 25796][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 25828][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[E][MLX90640:271]: MLX READING VALUE ERRORS
[ 30702][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[ 30703][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error 263
[logs (1).txt](https://github.com/Chill-Division/M5Stack-ESPHome/files/11360954/logs.1.txt)

ChillingSilence, 2024-06-11:
Looking at this @Cres09 , I see you have:
```  sda: 21  # I2C SDA PIN
  scl: 22  # I2C SCL PIN
```
But none of the M5Stack devices I'm familiar with use those on the Grove pins, so can you clarify what your ESP device is?

closed 2024-08-11 by ChillingSilence