Chill-Division/M5Stack-ESPHome #2: Esp32

donparlor, 2023-08-13:
Can this code be used with ESP32?

ChillingSilence, 2023-08-13:
Yes the Atom Lite, Atom S3 and PoESP32 are all ESP32's.

The yaml only tells ESPHome how to use the sensors, so you can even use them with an ESP8266 or something like an ESP32 Development Board.

closed 2023-08-13 by ChillingSilence

donparlor, 2023-08-14:
Sorry for re-opening this thread, but I would like to know if integrating MLX90640 with your ESPHome code I can expect to integrate it in Home Assistant and get a visual similar to this?
[image]

ChillingSilence, 2023-08-14:
Yes

donparlor, 2023-08-14:
Wonderful! Thank you @ChillingSilence! Your work with this code is great. I'll test it soon!

ChillingSilence, 2023-08-14:
Glad to hear it, happy to help