Chill-Division/M5Stack-ESPHome #3: (REQUEST) M5Stack I2C Hub 1 to 6 Expansion Unit (PCA9548APW)

waring192, 2024-02-22:
I dont don't suppose you are working on the following: M5Stack I2C Hub 1 to 6 Expansion Unit (PCA9548APW)

https://shop.m5stack.com/products/i2c-hub-1-to-6-expansion-unit-pca9548apw

ChillingSilence, 2024-02-22:
I'm not (yet) sorry, haven't had a need to purchase any of those recently. I'm presuming the older version is EOL'd? Thinking it may be worth keeping this issue open for now though coz eventually I likely will... Even if that's several months into the future 🤷‍♂️

jzucker2, 2025-05-05:
I got them working with this component https://esphome.io/components/tca9548a.html

Note: I was only able to get the arduino framework to be stable, the esp-idf version had some phantom tap issues for me.

ChillingSilence, 2025-05-05:
Do you have some example yaml you'd be willing to share @jzucker2 ?

jzucker2, 2025-05-05:
To reduce repetition, this should work with 3 files:

Main file like `mutiple_sensors_with_multiplexer.yaml` =>

```yaml
# Example
substitutions:
  node_name: multipletempsensors
  friendly_name: Multiple Temp Sensors
  area: Dev
  comment: "Temperature sensors using an I2C multiplexer"
  zeroth_sensor_name: "Top Shelf"
  first_sensor_name: "Middle Shelf"
  second_sensor_name: "Bottom Shelf"
  oversampling_value: "16x"
  update_interval_value: "60s"

esphome:
  name: '${node_name}'
  area: ${area}
  comment: ${comment}

logger:
  level: INFO

# Works with Gray Atom Lite
esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  - id: grove
    sda: GPIO26
    scl: GPIO32
    scan: true

# temp sensors

# https://esphome.io/components/tca9548a
tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: grove
    channels:
      - bus_id: multiplex0channel0
        channel: 0
      - bus_id: multiplex0channel1
        channel: 1
      - bus_id: multiplex0channel2
        channel: 2

sensor:
  - <<: !include {file: common/env_iv_sht4x_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel0", env_iv_sensor_name: "${zeroth_sensor_name}"}}
  - <<: !include {file: common/env_iv_bmp280_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel0", env_iv_sensor_name: "${zeroth_sensor_name}"}}
  - <<: !include {file: common/env_iv_sht4x_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel1", env_iv_sensor_name: "${first_sensor_name}"}}
  - <<: !include {file: common/env_iv_bmp280_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel1", env_iv_sensor_name: "${first_sensor_name}"}}
  - <<: !include {file: common/env_iv_sht4x_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel2", env_iv_sensor_name: "${second_sensor_name}"}}
  - <<: !include {file: common/env_iv_bmp280_sensor.yaml, vars: {env_iv_i2c_id: "multiplex0channel2", env_iv_sensor_name: "${second_sensor_name}"}}

```

Then a template file at `common/env_iv_sht4x_sensor.yaml` like:

```yaml
# https://shop.m5stack.com/products/env-iv-unit-with-temperature-humidity-air-pressure-sensor-sht40-bmp280
# https://esphome.io/components/sensor/sht4x.html
- platform: sht4x
  i2c_id: ${env_iv_i2c_id}
  address: "0x44"
  temperature:
    name: "${env_iv_sensor_name} Temperature"
  humidity:
    name: "${env_iv_sensor_name} Relative Humidity"
  update_interval: ${update_interval_value}

```

And one more for the other sensor at `common/env_iv_bmp280_sensor.yaml` like:

```yaml
# https://shop.m5stack.com/products/env-iv-unit-with-temperature-humidity-air-pressure-sensor-sht40-bmp280
# https://esphome.io/components/sensor/bmp280.html
- platform: bmp280_i2c
  i2c_id: ${env_iv_i2c_id}
  temperature:
    name: "${env_iv_sensor_name} Outside Temperature"
    oversampling: ${oversampling_value}
  pressure:
    name: "${env_iv_sensor_name} Outside Pressure"
  address: "0x76"
  update_interval: ${update_interval_value}

```

I have 3 or 4 of these set up around my house with 2-6 sensors attached to each.

And I take back what I said. This has been running fine on `esp-idf` and ran fine on `arduino` too

jzucker2, 2025-05-05:
Oh, I'm honored. I've been using your repo for a few years now. It's been really helpful! Thank you!

klim123123, 2025-05-06:
> Brilliant, thankyou! I don't have any of these units so I'll grab one or somebody else can verify and we'll get it up in the main list. Thanks again 👍

Just tested it on m5nanoc6 with "[Unit KMeter-ISO](https://shop.m5stack.com/products/kmeter-isolation-unit-with-thermocouple-temperature-sensor-max31855)". It's just works. 
Thank you.

```
i2c:
  - id: grove
    sda: GPIO2
    scl: GPIO1
    scan: true

tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: grove
    channels:
      - bus_id: multiplex0channel0
        channel: 0
      - bus_id: multiplex0channel1
        channel: 1
      - bus_id: multiplex0channel2
        channel: 2

sensor:
- platform: kmeteriso
  i2c_id: multiplex0channel0
  address: 0x67
  temperature:
    name: Temperature
  internal_temperature:
    name: Internal box temperature
```