Chill-Division/M5Stack-ESPHome #4: PBHub - Support for ADC

MRobi1, 2024-03-31:
Trying to run a pair of watering units off of a PB.Hub and would like to have the moisture values from the sensor.

When I set the pin to m5stack_pbhub it gives me an error of "[m5stack_pbhub] is an invalid option for [pin]."

```
sensor:
  - platform: adc
    pin:
      m5stack_pbhub: M5Stack_HUB
      # Use channel 0, pin number 0
      number: 01
      mode:
        input: true
```
Is this something that can be added in the m5stack_pbhub component to support adc?

ChillingSilence, 2024-06-11:
It's my understanding that the firmware was only created with the intention of it being a switch / binary sensor. It'd need further development done to take ADC which I haven't got capacity to fund development for sorry. Somebody else may add it though

regencape, 2024-10-31:
> Trying to run a pair of watering units off of a PB.Hub and would like to have the moisture values from the sensor.

This was exactly my plan. :) But somehow I run into an issue, not only with the sensors. ESPhome uploads the firmware without trouble, but there's an issue with the api. The api however seems to be configured just fine. When I run the firmware without the hub-part, the api gives no problem. I'm no expert, but tried to nail it down, and the problem appears when I introduce this part. I removed the sensor and switch part, for now.

```
 m5stack_pbhub:
   - id: 'M5Stack_HUB'
     address: 0x61
     sda: 16
     scl: 17
```

> INFO Starting log output from 192.168.xxx.xx using esphome API
> WARNING Can't connect to ESPHome API for xxx_some_name_xxx @ 192.168.xxx.xx: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.xxx.xx', port=6053))]: [Errno 111] Connect call failed ('192.168.xxx.xx', 6053) (SocketAPIError)

The original IP address is correct (otherwise the firmware wouldn't have updated anyhow). Any idea's? Are the pin numbers right? Can't find clear documentation (at least not clear to me) on the M5stack site to check these numbers. Help would be greatly appreciated.

ChillingSilence, 2024-10-31:
I'm sorry I'm not entirely too sure. PIN numbers will vary depending if you're using an Atom Lite, Nano C6, PoESP32 etc

What I'd suggest to help troubleshoot and show that the unit is up/alive is adding:
<pre>web_server: true
  port: 80</pre>

Because then you'll get a confirmation it's booting up or not, with the firmware modifications 

But realistically I would just have it plugged into the PC and use the serial output. You can always use the old esphome-flasher tool?

regencape, 2024-11-01:
Pinnumber is probably not the issue. I use the same setup as described (Atom Lite). It seems to be an issue with connecting to WiFi. When I try to connect I get the captive_portal. Trying to connect with the WiFi doesn't work either.  The log in the flasher tool gives: 

> [W][wifi_esp32:537][arduino_events]: Event: Disconnected ssid='wifi_here' bssid=[redacted] reason='Association Leave'. 

This gives me some clue on the problem. I found a similar issue on [GitHub](https://github.com/letscontrolit/ESPEasy/issues/1447) being an issue with DHCP server client lease time. See also same issue on [Hassio community](https://community.home-assistant.io/t/wifi-issues/407562). I'm gonna try to fix this some other time. Anyhow, thanks for your help how to troubleshoot better using the web_server --- without the true :) --- and the [flasher tool](https://web.esphome.io/). It gave the information I lacked.

regencape, 2024-11-01:
Never mind! It was the PIN number issue. I was confused since it was noted that it worked on Atom Lite and not on PoESP32. The PIN numbers in de yaml however are PoESP32. Maybe I still don't understand the logic, but I change it as described in the readme, and it's fixed. :)

ChillingSilence, 2024-11-03:
Closing due to being coopted, pull request for adc support can be made later if it happens :)

closed 2024-11-03 by ChillingSilence