Chill-Division/M5Stack-ESPHome #5: Integrating `thermal-camera` with

maovidal, 2024-05-31:
Hi!

I would like to be able to visualize in `Home Assistant` the image that the `thermal camera` can produce.

There is a custom `Home Assistant` component developed for this:
https://github.com/TheRealWaldo/thermal

It seems all it requires is a JSON struct with the data of the sensor. As a reference, there is already a development for a compatible hardware: https://github.com/pixelsquared/thermal_vision-ESP32-MLX90640/blob/main/src/main.cpp

I'm wondering if you would like to get a proposal to merge that functionality to your repo. If you have any considerations, please, let me know. I would gladly help with this.

ChillingSilence, 2024-05-31:
Hi @maovidal that sounds super helpful if you want to make a PR then I'd be keen as for that functionality :) Thanks!

plipof, 2025-01-18:
If I can help please let me know. I have M5 Thermal MLX90640 version 1 and 2 and M5StackAtom. Would love to have this ability!!

plipof, 2025-01-18:
and I don't know if this helps but this repository has functioning streaming and JSON. https://github.com/m5stack/M5StickC-Plus-TLite-FW?search=1

There seems to be a small error with the code for @realwaldo allowing his code to work with the TLite Code.