Chill-Division/M5Stack-ESPHome #7: PBHub - Support for Sensor

FunkeyMonkey, 2024-06-16:
Hey Chill, first thanks so much for putting all this out for everyone! Question on the Pb.hub code, i see you list option to use switch or binary sensor, is it possible to hook up one of the sht4x sensors to the hub? Having an issue trying to combine your sensor code to work with the format of the phub. Maybe i'm just missing something?

```
i2c:
  sda: 26
  scl: 32
  scan: true
  id: bus_1

sensor:
  - platform: sht4x
    temperature:
      name: "Temperature - SHT40"
      id: "temperature_sht40"
    humidity:
      name: "Humidity - SHT40"
      id: "humidity_sht40"
    address: 0x44
    update_interval: 10s
  - platform: bmp280
    temperature:
      name: "Temperature - BMP280"
      id: "temperature_bmp280"
      oversampling: 16x
    pressure:
      name: "Pressure BMP280"
      id: "pressure_bmp280"
    address: 0x76
    update_interval: 10s 

#####################################################################

# Link the m5stack_pbhub to a  binary sensor
binary_sensor:
  - platform: gpio
    name: "Dual Button on PBHUB Channel #2 Pin #0"
    pin:
      m5stack_pbhub: M5Stack_HUB
      # Use channel 2, pin number 0
      number: 20
      mode:
        output: true
      inverted: false
  - platform: gpio
    name: "Dual Button on PBHUB Channel #2 Pin #1"
    pin:
      m5stack_pbhub: M5Stack_HUB
      #Use channel 2, pin number 1
      number: 21
      mode:
        output: true
      inverted: false

ChillingSilence, 2024-06-16:
Glad to hear it's appreciated 👍 

As for what you're wanting to do... sort of. Those units you linked are designed for an analog device to be attached on their Grove ports. You probably want something like this:
https://shop.m5stack.com/products/mini-hub-module

Or like this:
https://shop.m5stack.com/products/i2c-hub-1-to-6-expansion-unit-pca9548apw

Because the sht4x uses i2c.

I haven't tried mixing and matching a pb.hub into a pa.hub or anything sorry so can't confirm if that will work out of the box if you're wanting to use both an i2c device as well as a gpio switch for example.

FunkeyMonkey, 2024-06-16:
oh darn, I had asked on esphome discord and someone directed me to the pb.hub one. I guess what's required to run 2 ENV.IV sht4 sensors off a atom/pwm combo? I do have one of the mini hubs but was told i couldn't run 2 of the same devices off it.

I have 2 Atoms, 2 pwm modules, 2 rs485 to ttl units an i2c ultrasonic sensor and 2 sht4 sensors. Just trying to find a way to split them up and run off the 2 atom/pwm modules

ChillingSilence, 2024-06-16:
When I'm doing something like that, I'll usually order an scd4x and an env.iii or env.iv and have them run off the mini 1-to-3 hubs because their i2c address is different.
For that reason, out of the box, the ultrasonic and and sht4x will run off the mini hub. If I was looking after a garden (assuming you are because of the choice in collection of sensors) I'd be looking for a sensor with CO2 anyways.

Apparently you can change the i2c address that your devices use, in sure that I've seen that referenced on the M5Stack site, but I've not tested it.

As for the rest, like the RS485-to-TTL, I'm not aware of an easy way to make multiple units like that run off the same ESP controller. The pb.hub would be so you could run a bunch of the "Dual button" devices, or even their watering pumps and control them over gpio separately.

FunkeyMonkey, 2024-06-16:
Yes, an indoor garden. So sounds like i'll need a [pa.hub](https://shop.m5stack.com/products/i2c-hub-1-to-6-expansion-unit-pca9548apw) and can run 2 env.iv sensors and the ultrasonic off the same atom and then for the rs485, what would i use to hook them up since they are UART? I can split them and have 1 on each atom?

ChillingSilence, 2024-06-16:
Yep that's what I've done, split them, but I'm using a PoESP32. You should be able to run the PWM (I think, depending on if it uses i2c) off the same as your env.iv, and an ultrasonic, but the rs485 needs it's own as far as I'm aware.

If somebody has told you that they can all run off the one, it may pay to ask them how, and if you find out lemme know 😊

FunkeyMonkey, 2024-06-16:
Haha gotcha. I've read so much stuff over the last few weeks i'm sure it's all just blended together. Here's the [pwm](https://shop.m5stack.com/products/atomic-pwm-base-fdd8447l) I have, my Atom just sits in it. Maybe i'm wrong but had assumed i could use the Atom like normal (with a hub) and still use the pwm function alongside it.

ChillingSilence, 2024-06-16:
Ah, yes that Base will work because it uses the other pins.

Good luck!

FunkeyMonkey, 2024-06-17:
> When I'm doing something like that, I'll usually order an scd4x and an env.iii or env.iv and have them run off the mini 1-to-3 hubs because their i2c address is different. For that reason, out of [...]

Oh speaking of different i2c address, how would you list that for both of them to work in the same config using one of the mini hubs? If I try and list something like this, i get an error that i2c is duplicate.

```
i2c:
  sda: 26
  scl: 32
  scan: true
  id: bus_1

sensor:
  - platform: sht4x
    temperature:
      name: "Temperature - SHT40"
      id: "temperature_sht40"
    humidity:
      name: "Humidity - SHT40"
      id: "humidity_sht40"
    address: 0x44
    update_interval: 10s
  - platform: bmp280
    temperature:
      name: "Temperature - BMP280"
      id: "temperature_bmp280"
      oversampling: 16x
    pressure:
      name: "Pressure BMP280"
      id: "pressure_bmp280"
    address: 0x76
    update_interval: 10s

i2c:
  sda: 16
  scl: 17
  scan: true
  id: bus_1

 - platform: sonic_i2c
   i2c_id: bus_1
   address: 0x57
   name: "Ultrasonic Sensor 1"
   id: ultrasonic1
   unit_of_measurement: mm
   update_interval: 5s
```

ChillingSilence, 2024-06-17:
The i2c pins don't change so you only need one of those Config entries. The "address" is in each individual sensor component

FunkeyMonkey, 2024-06-17:
Oh duh I see that now. I was thinking the 16/17 and 26/32 were the pins/ports used. I now see it's just the atom vs Poe pins. I get it now, as long as the address isn't a duplicate I can hook up multiple sensors. All this time I was thinking it was the pins 🤦 Thanks again you're a rockstar

ChillingSilence, 2024-06-17:
Correct ✌️

closed 2024-06-17 by ChillingSilence