Chill-Division/M5Stack-ESPHome #8: RS485 to ttl & THC-S sensor

FunkeyMonkey, 2024-06-16:
Sorry having couple issues again. Getting the following errors. I have the 4 THC-S wires hooked up as follows: black on gnd, brown on 12v, yellow on A and blue on B. I have 2 RS485 modules and THC sensors and have tried mixing and matching both to try and rule out hardware.
 > WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.

> [D][modbus_controller:043]: Modbus command to device=1 register=0x00 countdown=0 no response received - removed from send queue

```
  libraries:
    - Wire
    - EspSoftwareSerial@8.1.0

external_components:
  - source: 
      type: local
      path: components
    components: [tdr_soil_sensor3]
 
### RS485
uart:
  id: modbus_serial
  rx_pin: 1
  tx_pin: 2
  baud_rate: 4800
  stop_bits: 1

modbus:
  id: modbus1
  uart_id: modbus_serial

modbus_controller:
  - id: con_modbus_name
    update_interval: 11s
    address: 0x1
    command_throttle: 150ms
    setup_priority: -10

sensor:
  - platform: modbus_controller
    name: "Moisture WC"
    id: WC
    address: 0x0000
    unit_of_measurement: "%"
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      lambda: return x / 10;
    state_class: measurement
  - platform: modbus_controller
    name: "Temp"
    id: C
    address: 0x0001
    unit_of_measurement: "°C"
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      lambda: return x / 10;
    state_class: measurement
  - platform: modbus_controller
    name: "Conductivity 1 dS/m = 1 mS/cm"
    id: EC
    address: 0x0002
    unit_of_measurement: "ms/cm"
    register_type: read
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      lambda: return x / 100;
    state_class: measurement

ChillingSilence, 2024-06-16:
Did you change the UART pins for your Atom? Coz 1&2 isn't for the Atom is it?

FunkeyMonkey, 2024-06-16:
oh geez thanks. I had tried 26 and 32 before but looks like i had them backwards. All working now. Thanks again!

ChillingSilence, 2024-06-17:
Glad you got it sorted. Maybe I should make some changes so the config is easy to uncomment or similar... I'll have a think 🤔

closed 2024-06-17 by ChillingSilence