Chill-Division/M5Stack-ESPHome #11: Using Mini 3A Relay Unit with ATOM Lite ESP32 - Boot loop

zibous, 2024-08-16:
Using Mini 3A Relay Unit with ATOM Lite ESP32 - Boot loop

When I do a simple test with ESPHome version 2024.7.3, it does not work. 
The ATOM Lite always starts and runs in boot mode until SAVE mode. 

``` yaml
## -----------------------------------------------------------------------------------------------------
## ATOM LITE S3 - ESP32-S3FN8
##
## Detecting chip type... ESP32
## Chip is ESP32-PICO-D4 (revision 1)
## Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
## Crystal is 40MHz
## -----------------------------------------------------------------------------------------------------
substitutions:

  ## device settings
  hostname: "atomlite"
  friendly_name: "Test"
  device_description: "Testcase: Relais"

  # The format of the name should be author_name.project_name.
  projectname: "Atom_lite.Testcase"
  appversion: "1.0.0"
  hardware: "ATOM LITE S3 - ESP32-S3FN8"
  location: "Vorratsraum"

  ## logger settings
  log_level: "INFO" 
  log_baudrate: "9600" 

  ## device pins and timeer
  relay_pin: "16"
  relay_on_duration: 2s
  
## ----------------------------------------------------------------
## Board ATOM LITE
## ----------------------------------------------------------------
esp32:
  board: m5stack-atom
  framework:
    type: arduino

## ----------------------------------------------------------------
## APPLICATION ESPHOME
## ----------------------------------------------------------------
esphome:
  name: ${hostname}
  build_path: ./build/${hostname}
  comment: ${device_description}
  project:
    name: ${projectname}
    version: ${appversion}
  area: ${location}
  name_add_mac_suffix: false
  min_version: 2024.6.6

## ------------------------------------------------------------------
## COMPONENT LOGGER
## https://esphome.io/components/logger.html
## ------------------------------------------------------------------
logger:

## ------------------------------------------------------------------
## COMPONENT WIFI
## https://esphome.io/components/wifi.html
## ------------------------------------------------------------------
wifi:

  networks:

    # default wifi smarthome iot
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
      priority: 100.00

    # fallback 1: wifi 2.4Ghz + 5GHz
    - ssid: !secret ssid2_name
      password: !secret ssid2_pswd
      priority: 50.0

    # fallback 2: wifi 2.4Ghz + 5GHz
    - ssid: !secret ssid3_name
      password: !secret ssid3_pswd
      priority: 30.0

    # fallback 3: wifi 2.4Ghz
    - ssid: !secret ssid4_name
      password: !secret ssid4_pswd
      priority: 10.0

  domain: !secret domain
  reboot_timeout: 5min
  enable_on_boot: true
  power_save_mode: NONE

## ------------------------------------------------------------------
## COMPONENT Over-the-Air Updates
## https://esphome.io/components/ota/#over-the-air-updates
## ------------------------------------------------------------------
ota:
  platform: esphome
  password: !secret ota_pswd

## ------------------------------------------------------------------
## COMPONENT WEBSERVER
## The web_server component creates a simple web server on the node
## that can be accessed through any browser and a simple REST API.
## https://esphome.io/components/web_server.html
## ------------------------------------------------------------------
web_server:
  
## ---------------------------------------------------
## COMPONENT Home Assistant API
## https://esphome.io/components/api.html
## ---------------------------------------------------
api:
  id: atlredht
  port: 6053
  reboot_timeout: 0s

## ----------------------------------------------------------------
## Timed switch for frontdoor opener
## time for off is 2s
## ----------------------------------------------------------------
switch:
 - platform: gpio
   pin: 16
   id: relay
   name: "Relais"
   icon: mdi:door-open
   on_turn_on:
   - logger.log:
       level: INFO
       tag: "system"
       format: "Relais on"
   - delay: ${relay_on_duration}
   - switch.turn_off: relay
   - logger.log:
       level: INFO
       tag: "system"
       format: "Relais off"

```

Used parts:
- ATOM LITE: https://docs.m5stack.com/en/core/atom_lite
- RELAY: https://docs.m5stack.com/en/unit/relay

Log: 
[logs_atom_lite_test_run.txt](https://github.com/user-attachments/files/16639857/logs_atom_lite_test_run.txt)

What is the workaround for this?
Thank you in advance for your help

ChillingSilence, 2024-08-16:
There looks to be some confusion between Atom Lite vs Atom Lite S3:
https://shop.m5stack.com/products/atoms3-lite-esp32s3-dev-kit

Two different units, with different settings, and different pinouts.
Is yours the white Atom Lite S3? If so, the pinout is wrong, see the README in the main repo.
Is it grey like the Atom Lite? If so, try changing your board. 

You can also clean your build files then flash without and mention of the relay at all, just get it built and online to ensure you've set that up correctly, then add the GPIO switch afterwards. That's how you'll know you have / haven't got the pins correct 😊

zibous, 2024-08-17:
Thanks.

Solved:  PIN 26

```
switch:
 - platform: gpio
   pin: 26
   id: relay
   name: "Relais"
  ----
```

closed 2024-08-17 by zibous