Chill-Division/M5Stack-ESPHome #13: (Working Example) M5Stack OLED Unit 1.3" 128 × 64 Display (SH1107)

DavidGeorge528, 2024-11-14:
Here is a minimal example for running the M5 Stack OLED Unit 1.3" display

```yaml
i2c: 
  sda: 2
  scl: 1

font:
  - file: "gfonts://Roboto@medium"
    id: my_font
    size: 18
    glyphs: '!"%()+,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/³µ'

display:
  - platform: ssd1306_i2c
    model: "SH1107 128x64"
    address: 0x3C
    update_interval: "30s"
    rotation: 90
    offset_y: 96
    lambda: |-
      it.printf(0, 0, id(my_font), "Hello");
      it.printf(0, 20, id(my_font), "World!");
```
[image]

This example assumes the device is plugged into the Grove port on an M5 stack Atom or similar, I2C pins will likely differ for others.

Some caveats:
- The `offset_y` was necessary (on my board at least) as without it the screen would display some debug information or even just garbage that I couldn't write over (see below)
[image]
- I get the following warning on each update, which I am not entirely sure how to fix or if its an issue at all
```
[13:26:33][W][component:237]: Component display took a long time for an operation (306 ms).
[13:26:33][W][component:238]: Components should block for at most 30 ms.
```

ChillingSilence, 2024-11-14:
Thanks @DavidGeorge528 ! I get that same warning message on my other screens too, it just takes too long for ESPHome to draw the screen from what I understand so that's not really too concerning.

Really appreciate you submitting this, I think it was the offset that may have been my issue because I was getting garbage on-screen too. I'll find my unit and give it a whirl on an Atom Lite and a PoESP32 later on.
Thanks again 🙏

ChillingSilence, 2026-09-09:
Finally got around to adding this, sorry, that was on me. Thankyou again @DavidGeorge528 🙏

closed 2026-09-09 by ChillingSilence