Chill-Division/M5Stack-ESPHome #17: ESPHome 2025.2.0 Config with PbHub fails to compile

danleongjy, 2025-02-23:
Hi, thanks for the work on this external component!  I'm using a PbHub in this way:
```
M5CameraF <---> PbHub <--|--> Unit Button
                         |--> Unit PIR
```
The full config for this device:
```
esphome:
  name: main-door-camera
  friendly_name: Main Door Camera

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret main_door_camera_api_key

ota:
  - platform: esphome
    password: !secret main_door_camera_ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Main-Door-Camera"
    password: !secret ap_password

  manual_ip:
    static_ip: 192.168.18.58
    gateway: 192.168.18.1
    subnet: 255.255.255.0

captive_portal:
    
external_components:
  - source:
      type: git
      url: https://github.com/Chill-Division/M5Stack-ESPHome/
    components: m5stack_pbhub

i2c:
  id: grove
  sda: 4
  scl: 13

m5stack_pbhub:
  - id: pbhub
    address: 0x61
    sda: 4
    scl: 13

binary_sensor:
  - platform: gpio
    id: motion
    name: Motion
    pin: 
      m5stack_pbhub: pbhub
      number: 31
      mode:
        output: true
    device_class: motion
    publish_initial_state: true
    filters:
      - delayed_off: 20s
    on_press: 
      then:
        - light.turn_on: 
            id: led
            effect: pulse
    on_release: 
      then:
        - light.turn_off: led
  - platform: gpio
    id: button
    name: Button
    pin: 
      m5stack_pbhub: pbhub
      number: 11
      mode:
        input: true
    publish_initial_state: true
    filters:
      - invert:

output:
  - platform: ledc
    pin: 14
    inverted: true
    id: led_output

light:
  - platform: monochromatic
    output:  led_output
    name: LED
    id: led
    effects:
      - pulse:
          name: "Pulse"
          transition_length: 1000ms
          update_interval: 1000ms
          min_brightness: 10%
          max_brightness: 100%

esp32_camera:
  id: camera
  external_clock:
    pin: GPIO27
    frequency: 20MHz
  i2c_pins:
    sda: GPIO22
    scl: GPIO23
  data_pins: [GPIO32, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO25
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15
  max_framerate: 2 fps
  jpeg_quality: 20
  brightness: 2

esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot
```

This successfully worked under 2025.1, but upgrading to 2025.2 caused the compile to fail:

```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/main-door-camera.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing main-door-camera (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- Wire @ 2.0.0
Compiling .pioenvs/main-door-camera/src/main.cpp.o
Archiving .pioenvs/main-door-camera/lib01c/libFS.a
Compiling .pioenvs/main-door-camera/libbc6/Update/Updater.cpp.o
Compiling .pioenvs/main-door-camera/lib22e/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
Compiling .pioenvs/main-door-camera/lib22e/ESPAsyncWebServer-esphome/AsyncWebSocket.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:317:69: error: invalid new-expression of abstract class type 'esphome::m5stack_pbhub::PBHUBGPIOPin'
   m5stack_pbhub_pbhubgpiopin_id_2 = new m5stack_pbhub::PBHUBGPIOPin();
                                                                     ^
In file included from src/esphome.h:51,
                 from src/main.cpp:3:
src/esphome/components/m5stack_pbhub/m5stack_pbhub.h:47:7: note:   because the following virtual functions are pure within 'esphome::m5stack_pbhub::PBHUBGPIOPin':
 class PBHUBGPIOPin : public GPIOPin {
       ^~~~~~~~~~~~
In file included from src/esphome/core/hal.h:4,
                 from src/esphome/components/light/light_transformer.h:4,
                 from src/esphome/components/light/light_state.h:11,
                 from src/esphome/core/controller.h:11,
                 from src/esphome/components/api/api_server.h:11,
                 from src/esphome/components/api/api_connection.h:8,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/gpio.h:61:23: note: 	'virtual esphome::gpio::Flags esphome::GPIOPin::get_flags() const'
   virtual gpio::Flags get_flags() const = 0;
                       ^~~~~~~~~
src/main.cpp:510:67: error: invalid new-expression of abstract class type 'esphome::m5stack_pbhub::PBHUBGPIOPin'
   m5stack_pbhub_pbhubgpiopin_id = new m5stack_pbhub::PBHUBGPIOPin();
                                                                   ^
Archiving .pioenvs/main-door-camera/libbc6/libUpdate.a
Compiling .pioenvs/main-door-camera/lib22e/ESPAsyncWebServer-esphome/WebAuthentication.cpp.o
*** [.pioenvs/main-door-camera/src/main.cpp.o] Error 1
========================== [FAILED] Took 9.09 seconds ==========================
```
For now, the device still works with the old config but I cannot upgrade it to 2025.2.  Please let me know if I can get more info to help with this issue.

danleongjy, 2025-02-28:
Flipping through the changelog for ESPHome 2025.2, I'm wondering if the error is related to [this change](https://github.com/esphome/esphome/pull/8151)?

MRobi1, 2025-03-15:
Hey! Just wondering if you've figured this one out?

ChillingSilence, 2025-03-15:
I've not been able to commit any time to this, I don't actually have any of those devices any longer to test with either. If there's somebody who can assist with correcting the code would be great thanks

danleongjy, 2025-03-21:
I'm willing to try to modify the files, but I have no idea how to programme in C!  I tried to follow the thread of changes in the ESPHome codebase:
* Changes to make `get_flags` mandatory in [#8151](https://github.com/esphome/esphome/pull/8151)
* Changes to other components in ESPHome in response to the above in [#8182](https://github.com/esphome/esphome/pull/8182)
* Based on #8182's [files changed](https://github.com/esphome/esphome/pull/8182/files), it sounds like all that is needed is to add `gpio::Flags get_flags() const override { return this->flags_; }` before `protected:` in any `.h` file.

So I'm guessing that I should insert the new code [here](https://github.com/Chill-Division/M5Stack-ESPHome/blob/83c0240ae7c311d8d0df20472352623fe17ab8f4/components/m5stack_pbhub/m5stack_pbhub.h#L59) for the PbHub?

danleongjy, 2025-03-22:
Holy crap. It worked. =)  Will write up a PR for it when I have the time, but the change described above was effective and I was able to compile for ESPHome 2025.2.1.

MRobi1, 2025-03-22:
Nice!! Great work!

PR #20 (2025-03-23): Update M5Stack PbHub to work with ESPHome 2025.2 GPIOPin get_flags
Addresses [#8151](https://github.com/esphome/esphome/pull/8151) as suggested in #17.  This allows ESPHome configuration to compile with this external component on 2025.2.  Closes #17.
closed 2025-03-23 by ChillingSilence